Sonic-Amiga/ESP8266-Faikin #4: Confuses Auto vs Quiet modes

Sonic-Amiga, 2023-06-21:
In both cases (when activated by RC) Faikin reports fan speed selector as "auto".
Setting these modes also needs verification

revk, 2023-06-23:
Good luck with that, I failed to find any indication of a difference if what comes from the a/c.

Sonic-Amiga, 2023-06-23:
@revk Just verified on my older ATX20K "Siesta" with original Daikin BRP controller; everything is good, a proper f_rate is reported. So it's possible. Perhaps your AC, on which you tested that, is buggy. Its firmware authors are also mere humans after all.

revk, 2023-06-24:
I am pretty sure my old Daikin worked too, in terms of f_rate, but I have yet to find the underlying S21 message for this.

Sonic-Amiga, 2024-07-26:
The issue still persists. I managed to order an original BRP box from old stock, should arrive here in a week, so i'll be able to experiment.

Sonic-Amiga, 2024-08-13:
Solved by today's commits.

closed 2024-08-13 by Sonic-Amiga