Sonic-Amiga/ESP8266-Faikin #7: Errors attempting make

jonchaka, 2023-07-24:
Please  let me know if doing incorrectly.
Have had no issues using make with other esp8266 projects.

Attempting to run 'make' or 'make menuconfig' resultsin the same error:

$ make
make: components/ESP32-RevK/buildsuffix: Command not found
Makefile:10: *** Please install /bin/csh or equivalent.  Stop.

$ make menuconfig
make: components/ESP32-RevK/buildsuffix: Command not found
Makefile:10: *** Please install /bin/csh or equivalent.  Stop.

Sonic-Amiga, 2023-09-14:
Hello! Sorry for missing out. Are you on mingw or Linux? Looks like csh is not installed.

Sonic-Amiga, 2024-07-29:
Closing due to old age. Please re-raise if more question ever pop up. A pity i missed the message :(

closed 2024-07-29 by Sonic-Amiga