Sonic-Amiga/ESP8266-Faikin #9: No wifi access point after flash bin file

manhst, 2023-09-27:
Hi my friend,

After I flash Faikin-8266.bin file into my nodemcu. 
Flash success but I can't see wifi access point. 
May I know what is problem?

Sonic-Amiga, 2024-07-29:
Not sure if you're still reading this. I managed to miss out lots of messages. Don't get emails from here.
If yes, my first question. Does the firmware start up at all ? Bootup process should produce some log on the serial port at 115200 8n1, the last message should be "trying S21", after that A/C comm takes over the serial.

srodighiero, 2024-08-02:
Same here, but not clear for me the flashing procedure. Is it enough with the nodemcu firmware programmer to just flash the Faikin-8266.bin?

Sonic-Amiga, 2024-08-02:
Hello! No, it's not enough. At least once you need to reflash all partitions and bootloader.
There are tons of fancy flashers available, i cannot give precise guide for all of them, sorry. My docs should provide esptool command line; you can see all parameters from there. Basically you need binary images and their start addresses.