Sonic-Amiga/ESP8266-Faikin #10: Problem with Daichi dw22-b

zKeepers, 2023-10-28:
HI. I have the same device - Daichi dw22-b ver-03 and i have a problem with connect it to computer. When connect it i get an error:
```shell
sudo dmesg
[ 2940.865999] usb 1-2-port2: Cannot enable. Maybe the USB cable is bad?
[ 2941.837116] usb 1-2-port2: Cannot enable. Maybe the USB cable is bad?
[ 2941.837587] usb 1-2-port2: attempt power cycle
[ 2943.126045] usb 1-2-port2: Cannot enable. Maybe the USB cable is bad?
[ 2944.098011] usb 1-2-port2: Cannot enable. Maybe the USB cable is bad?
[ 2944.099111] usb 1-2-port2: unable to enumerate USB device
``` 
So i can't get a COM port for flashing this device. Tried different positions of "UART" switch.
Does i need another device to connect dw22-b to computer?
When connected to a computer or air conditioner, dw22-b receives power, connects to Wi-Fi and works correctly with the cloud application.

Sonic-Amiga, 2024-07-03:
Hello! You're probably plugging a USB cable directly into the controller; and no; that's not how it works. That USB connector is not a real USB; it's a 5V TTL serial. You need an adapter like this one https://www.waveshare.com/pl2303-usb-uart-board-micro.htm .

You can see Daichi pinout from the schematic i provided.

Sonic-Amiga, 2024-07-29:
HW docs are updated and improved, connection procedure is documented with a photo. Closing this ticket. Please re-raise for more questions.

closed 2024-07-29 by Sonic-Amiga