Sonic-Amiga/ESP8266-Faikin #11: x50a pins

Poopikakas, 2023-11-09:
Hi, is there a pinout description for the esp8266 regarding the x50a cable.

Sonic-Amiga, 2024-07-03:
Hi! Sorry for so-o-o-o delayed reply.
Unfortunately i don't have any info about cables; frankly speaking i have never seen x50a A/C. You may ask here https://github.com/revk/ESP32-Faikin

Sonic-Amiga, 2024-11-04:
Recently i've acquired a BRP069C41 controller, which is designed for X50 protocol. Here are photos of AC-side connectors and their pins:
[image]
[image]

Sonic-Amiga, 2024-11-05:
I hope this answers the question. Please reopen if required

closed 2024-11-05 by Sonic-Amiga