Sonic-Amiga/ESP8266-Faikin #12: GPIO Pins for UART

ppuetsch, 2024-02-08:
Hi @Sonic-Amiga, 

thanks a lot for your work. 
I do have some D1-Mini ESP8266 Boards lying around and plan to use one to monitor/control my Daikin AC Unit. 
I am wondering which ESP-Pins were used for the UART-Communication? I tried to figure that information out from the code, but was unfortunately not successful. Could you hint me at the specific definition, or just provide the GPIO-Numbers that are used?

Best regards, 
Philip

Sonic-Amiga, 2024-02-08:
Hello, nice to meet you, welcome to the club.
I am not at home, can't recall numbers, but you may check the reference schematic in Hardware/. It will tell you everything.

Remember that A/C port uses 5V for logic and provides 14V for power!!! Do not connect the ESP directly!

Sonic-Amiga, 2024-02-09:
Checked the code. Rx is GPIO3 and Tx is GPIO1.

In fact these can't be switched. 8266 only has two UARTs; and UART1 doesn't have a receiver, only transmitter, so it's only useful for dumping logs. Faikin logs to mqtt instead, so there's no need to bother.

ferbulous, 2024-06-22:
Hi @Sonic-Amiga 
I’ve flashed faikin-esp8266 to my d1 mini
Do I still need to configure the pins for tx & rx in the advanced settings? I’m still using the default setting but it’s still system offline 
Also, my daikin has CN_WIR port,  this is how I’ve wired CN_WIR to d1 mini. 

[image]

i thought the bottom pin was 12v, but it’s actually 5v

Sonic-Amiga, 2024-06-23:
Hello! Nice to meet you
Yes, CNW port uses 5V levels; and I/O is also 5V. You need a level converter.
Also refer to experience by another user. I don't recall default settings of my binary build, could be that CN-WIRED protocol is disabled by default. Please check advanced settings.
 I am attacked by laziness, haven't verified my latest version yet, so haven't released it. Will hopefully do this week.

Sonic-Amiga, 2024-06-23:
NB: rx/tx pin in advanced settings don't do anything in my port, they are ignored. Pins aren't reprogrammable for 8266, so rx = 3 and tx = 1 are the only valid choice.

ferbulous, 2024-06-24:
Thanks @Sonic-Amiga 
I checked using multimeter and the bottom pin is indeed 5v and sufficient to power the d1 mini without level converter
I’ve got it online now after some tips from here and swapping the tx/rx cable (a/c model FTN15PV1L)

https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/14#issuecomment-2185014650

so the pinout from the cn_wir port should be like this

[image]

Only issue I’m having is adjusting temperature sometimes doesn’t seem to work. For example If i lower to 24 C from 25 C, it doesn’t do anything (toggle goes back to 24 C) but increasing to 26 from 25 C works fine. I might need to raise github issue for this

Sonic-Amiga, 2024-06-24:
> sufficient to power the d1 mini without level converter

I mean 5<>3.3 converter on data lines. At least officially ESPs aren't 5V-tolerant as far as i know. You may eventually get it fried.

> (toggle goes back to 25 C) 

Hm, interesting... I'll check, however i didn't have any problems like this at least on simulator. There's really nothing special about 25 or 24 C... And AC only reports settings change when you use the remote.

ferbulous, 2024-06-24:
> I mean 5<>3.3 converter on data lines. At least officially ESPs aren't 5V-tolerant as far as i know. You may eventually get it fried

I see, so I need to add level shifters for tx/rx pins

> Hm, interesting... I'll check, however i didn't have any problems like this at least on simulator. There's really nothing special about 25 or 24 C..

Not just 24 C, if I pick specific temperature like from 25 to 21 for example, the sliders reverts back to 25. How I do view or dump the logs here when that happens?

Sonic-Amiga, 2024-06-26:
>  I'm not sure how to get the logs yet

Just go to "advanced settings" and enable "Debug". Log is sent to MQTT, topic names are <level>/<controller_name>/...
My faviorite way to read all of them and discard the rest is:

mosquitto_sub -v -t # |grep faikin_test

(faikin_test) is a name of my development installation.

I checked my own installation, but the problem is that i don't have HomeAssistant. I am using OpenHab, and it uses proprietary Daikin native http-based API; Faikin implements that too.

In fact, as i said, the protocol is very dumb, there's no feedback from the AC, so even if the AC dropped the command, the interface should still reflect the new state. There's no way to query the AC for its status, it only sends updates about indoors temperature and nothing more. The only way to make the AC to send its current state is to use its IR remote; in this case it sends an update about its new state.

So probably it's a higher-level bug.

During the recent time i pulled in lots of changes from RevK's upstream. I made a fresh betarelease today; please check, many things have been fixed there.

Sonic-Amiga, 2024-07-27:
So, i created a Wemos-D1Mini folded in Hardware/ in my repo and saved the information there. I am closing this issue; please re-open or create new if there are some issues

closed 2024-07-27 by Sonic-Amiga

Sonic-Amiga, 2025-04-13:
Hi! ESP is +5V tolerant, so it appears receiving data is no problem. As to tx, it depends on the A/C. People report success with CN_WIRED, on my S21 unit direct connection didn't work. The A/C doesn't "hear" a +3V signal. I built this simple adapter https://github.com/Sonic-Amiga/ESP8266-Faikin/tree/main/Hardware/S21Adapter to circumvent the issue. You need to configure tx inversion to use it.
 I got a bit tired from this project, so put it aside for the time being. But it still runs great in my home.