Sonic-Amiga/ESP8266-Faikin #14: CN_WIRED and offline

fisiomacao, 2024-06-09:
[image]
Hello.
I get this image with "system is off line" intermittently, it changes every second. And can't do anything with it. Help? Or can someone guide me to throubleshoot this?
Thanks.

fisiomacao, 2024-06-09:
With mqtt explorer I got this:
`{"protocol":"CN_WIRED","online":false,"autor":10.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}`

lank23, 2024-06-16:
I have the same issue, but the web interface updates went key presses from remote are received.  I put my scope on the RX and TX lines , RX is getting input from the CN_WIRED port on the unit, but nothing is ever send out on the TX line of the ESP32.  Changed to different GPO, different setting in advance page, same results...how to correct?

ferbulous, 2024-06-22:
Hi @fisiomacao did you have any progress with this? I’m also just getting system offline

lank23, 2024-06-23:
Ok so I figured maybe my esp32 device was bad or non compatible in some way, so I switched to a ESP8286 device "ESP8266 ESP-12 ESP-12F NodeMcu Mini D1 Module WeMos Lua 4M Bytes" and uploaded set TX = 1 RX = 3, selected only CN-WIRE for protocol and it is working! So not sure if its the hardware or the firmware...Thanks

ferbulous, 2024-06-23:
> uploaded set TX = 1 RX = 3, selected only CN-WIRE for protocol

Hi, how do you change tx & rx? Do you just change it in the advanced settings?

[image]

lank23, 2024-06-23:
Yes I changed them in the advanced settings, once it connected the option to select different protocols is gone and just a text box to enter protocol number, I did not enter the "2" in the protocol box, guess it is set when protocol is auto found .  I also turned off the switching of the TX/RX GPIO options along with the other protocols. see pic.
[image]
[image]

fisiomacao, 2024-06-30:
So, I'm trying to use it with a wemos d1 mini. I used pins 5V, GND, RX and TX (TX from AC connecting to RX in ESP, RX to TX). It was causing this issue. I read somewere else that UART pins weren't those, and read I should use GPIO2 for RX and GPIO4 for TX. It became online, but doesn't read anything nor send info to the device.

lank23, 2024-06-30:
wemos d1 mini is the same type of board I am using.  TX =1 and RX = 3 should make it connect.  Also the TX/RX pins might be hard-coded in the firmware due the chip has no other option so the web setting probably does nothing...I had some issues before but when I re-flashed the unit with a full erase it worked.

Full erase command
`esptool.py  erase_flash`

Program command
`esptool.py --port /dev/ttyUSB0 write_flash 0x0 Faikin-8266-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-8266.bin`

Sonic-Amiga, 2024-07-02:
> so the web setting probably does nothing

Yes, you're absolutely correct, and i even removed it from recent version (under testing, not released yet). Pins aren't reprogrammable; and even uart1 isn't usable because it doesn't have rx; so the only option is uart0 on pins 1 and 3.

Make sure that cn_wired protocol is enabled. Setting could default to 0 from old version.

Sonic-Amiga, 2024-07-03:
If the problem persists; then:
1. Enable debug in advanced settings.
2. Connect to MQTT
3. Collect some logs as described here https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/12#issuecomment-2192696938

LowKey88, 2024-07-04:
Hi @Sonic-Amiga,

I am encountering an same issue (intermittent system status offline/online) with my ESP8266 Faikin setup. Below are the details of my configuration and the problem I am facing:

Configuration:

I have configured my device with the following MQTT settings:

	•	Device ID: 84CCA885AA1B (Faikin-8266: 57fee20)
	•	MQTT Topics and Payloads:

`homeassistant/climate/84CCA885AA1B/config {
  "unique_id": "84CCA885AA1B",
  "dev": {
    "ids": ["84CCA885AA1B"],
    "name": "84CCA885AA1B",
    "sw": "57fee20",
    "mf": "RevK",
    "cu": "http://84CCA885AA1B.local/"
  },
  "icon": "mdi:thermostat",
  "~": "command/84CCA885AA1B",
  "min_temp": 16,
  "max_temp": 32,
  "temp_unit": "C",
  "temp_step": 1,
  "temp_cmd_t": "~/temp",
  "temp_stat_t": "84CCA885AA1B",
  "temp_stat_tpl": "{{value_json.target}}",
  "curr_temp_t": "84CCA885AA1B",
  "curr_temp_tpl": "{{value_json.temp}}",
  "mode_cmd_t": "~/mode",
  "mode_stat_t": "84CCA885AA1B",
  "mode_stat_tpl": "{{value_json.mode}}",
  "fan_mode_cmd_t": "~/fan",
  "fan_mode_stat_t": "84CCA885AA1B",
  "fan_mode_stat_tpl": "{{value_json.fan}}",
  "swing_mode_cmd_t": "~/swing",
  "swing_mode_stat_t": "84CCA885AA1B",
  "swing_mode_stat_tpl": "{{value_json.swing}}",
  "swing_modes": ["off", "V"],
  "pr_mode_cmd_t": "~/preset",
  "pr_mode_stat_t": "84CCA885AA1B",
  "pr_mode_val_tpl": "{{value_json.preset}}",
  "pr_modes": ["eco", "boost", "home"]
}
`

However, I am receiving an error in some of the messages:

`error/84CCA885AA1B/comms {
  "protocol": "CN_WIRED",
  "badsum": "D2",
  "data": "0038000000000022"
}`

Despite this error, I am able to see the state updates and other information being published to the MQTT broker. Here are some of the state messages:

`state/84CCA885AA1B/status {
  "protocol": "CN_WIRED",
  "online": true,
  "home": 25.0,
  "heat": false,
  "power": false,
  "mode": "A",
  "temp": 20.0,
  "fan": "A",
  "swingv": false,
  "econo": false,
  "powerful": false,
  "autor": 0.0,
  "autot": 0.0,
  "auto0": "00:00",
  "auto1": "00:00",
  "autop": false
}`

I have verified the MQTT messages and confirmed that they are received correctly in Home Assistant.

Sonic-Amiga, 2024-07-04:
Hello! I have updated the file in "release" directory, please test. Your problem should be solved.

Your conditioner sends some additional data in new format, which Faikin did not understand.
Two interesting questions:
1. What's your AC model ?
2. The unknown packet has number 38 in it. Any idea what that could mean ? Is there any setting on your AC set to 38 ? We are interested because we still don't know how to interpret that data; currently it will be simply ignored.

Unfortunately i don't have my own public update server. I push releases to RevK so that they are available at revk.uk, but that comes with a delay; plus i'd like them to be tested. You can roll your own server in order to use OTA; or you can reflash the board via serial port if you aren't lazy. Sorry for that.

LowKey88, 2024-07-05:
Hi @Sonic-Amiga

I’ve tested the updated file and it’s working perfectly. My AC model is FTKG50T. I’ve also hosted your firmware on my web hosting, which can be accessed here: http://faikin-ota.home-assistant.my

Thank you for the update!

Sonic-Amiga, 2024-07-05:
Thank you for support and testing. I'm happy my work is useful for someone else.
I'll try backporting my code to ESP32 version; it should definitely work better.

Sonic-Amiga, 2024-07-09:
BTW, how reliable is your server ? Can i make it my official update path ? In such case i'd need some access to this dir to be able to upload files.

LowKey88, 2024-07-10:
I can ensure it’s reliable! 😊 Sure, I’m happy with that. You can make it your official update path.

Please let me know how you would like me to provide access to the directory for uploading files, and I’ll set it up for you.

Sonic-Amiga, 2024-07-10:
I don't care. Could be FTP, WebDav, SCP, whatever that would work for you best.

type-rke, 2024-07-11:
sorry for hijacking this topic, but i wanted to upgrade, but is is not working?

i receive upgrade failed
[image]

Sonic-Amiga, 2024-07-11:
> i have the esp32 faikin board

Just paid attention to that. :)

Sonic-Amiga, 2024-07-30:
@LowKey88 Hello again!

Three weeks ago we've been talking about setting up an update server, and you silenced.
My last reply was https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/14#issuecomment-2219600625

LowKey88, 2024-07-30:
@Sonic-Amiga send me your email. I will send out the credentials.

Sonic-Amiga, 2024-07-31:
pavel_fedin <at> mail.ru

LowKey88, 2024-07-31:
OK done

Sonic-Amiga, 2024-08-04:
Everything works fine, closing.

closed 2024-08-04 by Sonic-Amiga