Sonic-Amiga/ESP8266-Faikin #15: AC Control intermittenly not sent/received (CN_WIRED protocol)

ferbulous, 2024-06-25:
Details:
Port : CN_WIR
Model: FTN15PV1L

Hi, continuing from my previous comment.

https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/12#issuecomment-2187535480

The ac control doesn't seem to be received by the ac initially and I have to keep resending the control  (2x -3x times)
I've enabled & disabled 'debug' option in the advanced setting but doesn't seem to make a difference.
I tried obtaining the logs by subscring to this topic ``èrror/pongroomac`` but I'm not getting any entries or maybe this isn't the correct topic for it and the ``state/pongroomac/status`` only list the following actions that were done when testing it

```
state/pongroomac/status,{"protocol":"CN_WIRED","online":true,"home":30.0,"heat":false,"power":true,"mode":"C","temp":23.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
state/pongroomac/status,{"protocol":"CN_WIRED","online":true,"home":30.0,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
state/pongroomac/status,{"protocol":"CN_WIRED","online":true,"home":30.0,"heat":false,"power":true,"mode":"C","temp":23.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
state/pongroomac/status,{"protocol":"CN_WIRED","online":true,"home":30.0,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
state/pongroomac/status,{"protocol":"CN_WIRED","online":true,"home":30.0,"heat":false,"power":true,"mode":"C","temp":26.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
```

Is there a proper way of obtaining the necessary logs to troubleshoot this?

Sonic-Amiga, 2024-06-26:
Hello!
Very interesting discovery. I know that reference proprietary controller, from which i reverse engineered this protocol, actually does send a command 3 times. I didn't understand why because AC of my user worked great without that. Apparently, some models may miss packets.

CN_Wired is a ve-e-e-e-e-ry dumb protocol. No handshakes, no responses, nothing. The command just flies away and that's it.

Sonic-Amiga, 2024-06-26:
I'll implement this behavior.

Sonic-Amiga, 2024-06-26:
> I tried obtaining the logs by subscring to this topic èrror/pongroomac 

There is also info/<name> and "Debug" also enables debug/<name> . I'd use:

mosquitto_sub -v -t # |grep pongroomac

I. e. subscribe to everything and then filter what we want.
This is an example which i obtained from changing 20 -> 24 degrees:
```
info/faikin_test/rx {"protocol":"CN_WIRED","data":"23045000000000E0"} <= Received data; AC sends current room temperature once per second
info/faikin_test/rx {"protocol":"CN_WIRED","data":"23045000000000E0"}
info/faikin_test/rx {"protocol":"CN_WIRED","data":"23045000000000E0"}
info/faikin_test/rx {"protocol":"CN_WIRED","data":"23045000000000E0"}
info/faikin_test/tx {"protocol":"CN_WIRED","data":"24045018010010A0"} <= "tx" - this is transmitted data
state/faikin_test/status {"protocol":"CN_WIRED","online":true,"home":23.0,"heat":false,"power":false,"mode":"A","temp":24.0,"fan":"A","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false} <= New status
```
This may be higher-level error, not communication with the AC. Because again, there's no confirmation from the AC.
Packet loss on AC side should look something like: you change a setting, HA reflects the new status, but the AC simply ignores it; and you need to flip your setting back and forth again.

ferbulous, 2024-06-27:
Thanks @Sonic-Amiga , I’ll test with the beta version
How do i ota update to beta? I’ve enabled beta in the advanced settings but I’m just getting error ota check.
Is there a correct url for the beta firmware check?

Meantime, I will unplug the d1 mini and serial flash the beta files

ferbulous, 2024-06-27:
Some logs here. After beta update, homeassistant seems to increase the temperatue by 0.1 instead of 1 so I'll need to check this again, I change the temperature on the web page instead.

Turning on the ac (I sent on command twice)
```
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
command/pongroomac/mode cool
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/tx {"protocol":"CN_WIRED","data":"2404500202001040"}
state/pongroomac/status {"protocol":"CN_WIRED","online":true,"home":30.0,"heat":                                                                             false,"power":true,"mode":"C","temp":24.0,"fan":"5","swingv":false,"econo":false                                                                             ,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop                                                                             ":false}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
state/pongroomac/status {"protocol":"CN_WIRED","online":true,"home":30.0,"heat":                                                                             false,"power":false,"mode":"C","temp":24.0,"fan":"5","swingv":false,"econo":fals                                                                             e,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","auto                                                                             p":false}
Faikin/pongroomac {"protocol":"CN_WIRED","ts":"2024-06-27T03:01:00Z","online":tr                                                                             ue,"home":30.00,"heat":false,"power":0.02,"mode":"C","temp":24.00,"fan":"5","swi                                                                             ngv":false,"econo":false,"powerful":false}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000012020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
command/pongroomac/mode cool
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/tx {"protocol":"CN_WIRED","data":"2404500202001040"}
state/pongroomac/status {"protocol":"CN_WIRED","online":true,"home":30.0,"heat":                                                                             false,"power":true,"mode":"C","temp":24.0,"fan":"5","swingv":false,"econo":false                                                                             ,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop                                                                             ":false}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}

```

Change temperature to 25 (command sent 3 times)

```
homeassistant/climate/C4D8D5020A31/config {"unique_id":"C4D8D5020A31","dev":{"ids":["C4D8D5020A31"],"name":"pongroomac","sw":"9d648f0","mf":"RevK","cu":"http://pongroomac.local/"},"icon":"mdi:thermostat","~":"command/pongroomac","min_temp":16,"max_temp":32,"temp_unit":"C","temp_step":0.1,"temp_cmd_t":"~/temp","temp_stat_t":"C4D8D5020A31","temp_stat_tpl":"{{value_json.target}}","mode_cmd_t":"~/mode","mode_stat_t":"C4D8D5020A31","mode_stat_tpl":"{{value_json.mode}}","fan_mode_cmd_t":"~/fan","fan_mode_stat_t":"C4D8D5020A31","fan_mode_stat_tpl":"{{value_json.fan}}","swing_mode_cmd_t":"~/swing","swing_mode_stat_t":"C4D8D5020A31","swing_mode_stat_tpl":"{{va [... cut ...] autop":false}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"30C4000000000030"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
```

ferbulous, 2024-06-27:
> In order to be able to use fresh betas your only choice currently is to roll your own update server.

Not a problem, so just put all these files in a folder and ota feature would pick them up for update?

[image]

> Okay, i will check. I blindly picked up all changes from upstream, so could be.

Yeah, the temperature sliders (in homeassistant) just increase/decrease by 0.1 value of 1 it seems. For now I just slide it enough so it gets to 1 digit, I'm not quite sure if there's something else I can change in the homeassistant thermostat card for the default value

> Do you mean Faikin's own web UI; and you also observe the issue there?
On the faikin web UI, the temparature slider change works fine, just with HA.

But if you're referring to the intermittent control issues/status bounce, it's still happening in faikin web ui and HA

> From your log i see that something causes "status" to bounce back to previous value. And that cannot be received data from the AC; because AC does not send the status back. It's something else. I could not reproduce this effect locally, but i'll try to take a careful look, maybe add some more debug specifically to catch that, because only you're the lucky one :)

Ok, I'll provide more log output this time
I'll test it on my other Daikin AC, seems like even the newer one I just got last week (model 2023) that comes with built in wifi adapter is using CN_WIR port as well.

Sonic-Amiga, 2024-06-27:
HA config bug found and fixed. Upstream code apparently has it because nobody got cnw up and running.
I'll play a bit more, try to reproduce the main bug with bouncing state, then post a new beta.

> so just put all these files in a folder, host it and ota feature would pick them up for update?

Yes. You don't need all the files, just the app (Faikin-8266.bin).
I guess HomeAssistant has ability to host arbitrary files (OH does), so in fact that should be easy. Would be nice if you provide some quick guide how to do it, i'll add it to documentation.

Sonic-Amiga, 2024-06-27:
I looked at your log more carefully, and your AC does something that Glinka65's AC does not do. It indeed keeps sending current mode.

```
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
info/pongroomac/rx {"protocol":"CN_WIRED","data":"24000002020000B1"}
```
'1' in the end means it's mode; '0' stands for sensors (that reports room temperature)

Can you tell me how frequent these are ? Once per second or faster ?

Sonic-Amiga, 2024-06-27:
I have uploaded a new beta with HA step fixed, please check. Once confirmed, i'll publish a PR upstream.

ferbulous, 2024-06-28:
> Can you tell me how frequent these are ? Once per second or faster ?

Anyway I can add timestamps to the logs? That would probably be more accurate
For me it’s almost every second

> HA config bug found and fixed. Upstream code apparently has it because nobody got cnw up and running.

Thanks for the fix, will try the latest beta today and I’ll share the guide to host (i think I use docker for it)

ferbulous, 2024-06-28:
Hi @Sonic-Amiga 
I've installed the latest beta, the temperature slider's fixed and the control is down to two attempts only instead of 3 (second attempts succesfully send the command each time).
Btw, I get failed check upgrade when I host the files locally (192.168.2.199:8999) using simplehttp server, is there additional steps for hosting the files?

Here are the logs for it

```2024-06-28 22:52:56 homeassistant/climate/C4D8D5020A31/config {"unique_id":"C4D8D5020A31","dev":{"ids":["C4D8D5020A31"],"name":"pongroomac","sw":"79f8781","mf":"RevK","cu":"http://pongroomac.local/"},"icon":"mdi:thermostat","~":"command/pongroomac","min_temp":16,"max_temp":32,"temp_unit":"C","temp_step":1,"temp_cmd_t":"~/temp","temp_stat_t":"C4D8D5020A31","temp_stat_tpl":"{{value_json.target}}","mode_cmd_t":"~/mode","mode_stat_t":"C4D8D5020A31","mode_stat_tpl":"{{value_json.mode}}","fan_mode_cmd_t":"~/fan","fan_mode_stat_t":"C4D8D5020A31","fan_mode_stat_tpl":"{{value_json.fan}}","swing_mode_cmd_t":"~/swing","swing_mode_stat_t":"C4D8D5020A31","swing_mode_stat_tpl":"{{value_json.swing}}","swing_modes":["off","V"],"pr_mode_cmd_t":"~/preset","pr_mode_stat_t":"C4D8D5020A31","pr_mode_val_tpl":"{{value_json.preset}}","pr_modes":["eco","boost","home"]}
2024-06-28 22:52:57 state/pongroomac {"id":"C4D8D5020A31","up":4,"mqtt-up":1,"app":"Faikin","version":"79f8781","build-suffix":"-8266","build":"2024-06-28T00:52:13","flash":64,"rst":1,"mem":37580,"ssid":"Huawei tenet","bssid":"6C5AB05132E6","rssi":-69,"chan":10,"ipv4":"192.168.2.168"}
2024-06-28 22:52:57 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":false,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-28 22:52:57 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:52:57 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:52:57 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:52:58 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:52:59 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:00 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:01 Faikin/pongroomac {"protocol":"CN_WIRED","ts":"2024-06-28T14:53:00Z","online":true,"heat":false,"power":false,"mode":"C","temp":25.00,"fan":"5","swingv":false,"econo":false,"powerful":false}
2024-06-28 22:53:01 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:02 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:03 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:04 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:05 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:06 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:07 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:08 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:09 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:10 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:11 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:12 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:13 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:14 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:15 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:16 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:17 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:18 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:19 command/pongroomac/mode cool
2024-06-28 22:53:20 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:20 info/pongroomac/tx {"protocol":"CN_WIRED","data":"2504500202001050"}
2024-06-28 22:53:20 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-28 22:53:20 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:21 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":false,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-28 22:53:21 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:53:22 command/pongroomac/mode cool
2024-06-28 22:53:22 info/pongroomac/rx {"protocol":"CN_WIRED","data":"32C4000000000050"}
2024-06-28 22:53:23 info/pongroomac/tx {"protocol":"CN_WIRED","data":"2504500202001050"}
2024-06-28 22:53:23 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"home":32.0,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0, [... cut ...]
2024-06-28 22:59:00 Faikin/pongroomac {"protocol":"CN_WIRED","ts":"2024-06-28T14:59:00Z","online":true,"home":30.00,"heat":false,"power":false,"mode":"C","temp":25.00,"fan":"5","swingv":false,"econo":false,"powerful":false}
2024-06-28 22:59:01 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:59:02 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:59:03 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-28 22:59:04 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
```

Sonic-Amiga, 2024-06-28:
> the temperature slider's fixed

Thank you.

> control is down to two attempts only instead of 3 

Hm, i actually didn't change anything regarding protocol.

One more question. When you turn the AC on, the status bounces back to off, but does the AC actually turn on ? Does it receive the command ?

Thank you for logs, but nothing new there. I think i know what to do about it and we'll try. Need some time.

Sonic-Amiga, 2024-06-28:
> I get failed check upgrade when I host the files locally (192.168.2.199:8999) using simplehttp server

Hm, what i didn't try is to specify custom port. I am using default Apache install on my desktop. The file is hosted under htdocs/ota:
```
$ ls -l /c/Apache24/htdocs/
.htaccess   faikin/     index.html  ota/

Sonic@Sonic MINGW64 ~
$ ls -l /c/Apache24/htdocs/ota/
total 928
-rw-r--r-- 1 Sonic Отсутствует 950064 May  5 21:46 Faikin-8266.bin
```

I am able to see the file in browser at http://192.168.0.2/ota/ , and my "otahost" is set to 192.168.0.2/ota

BTW, this assumes "otabeta" is off. Beta would use a different file name i suppose; frankly speaking never checked for that. It's not me who codes that part.

Maybe URL parser chokes at port number; that's the only diff between yours and my setup i can see. You can also check your simplehttpd logs.

Sonic-Amiga, 2024-06-28:
Just looked at librevk code. If "otabeta" is activated, it will look in beta/ subdir. I. e. http://192.168.0.2/ota/beta/Faikin-8266.bin

Sonic-Amiga, 2024-06-28:
I have pushed one more beta, it now repeats packet being sent 3 times, the same as original HW does. Please test.

ferbulous, 2024-06-28:
> Just looked at librevk code. If "otabeta" is activated, it will look in beta/ subdir. I. e. http://192.168.0.2/ota/beta/Faikin-8266.bin

I'll try just using default 80 port this time

> I have pushed one more beta, it now repeats packet being sent 3 times, the same as original HW does. Please test.

Thanks, just installed it, looks like the same behavior with 2nd attempt only sends the command

> One more question. When you turn the AC on, the status bounces back to off, but does the AC actually turn on ? Does it receive the command ?

The AC only turns on (or cooling mode) on the 2nd attempt, just like with the temerature control

Here are the logs

```
2024-06-29 07:24:53 homeassistant/climate/C4D8D5020A31/config {"unique_id":"C4D8D5020A31","dev":{"ids":["C4D8D5020A31"],"name":"pongroomac","sw":"31e0ca7","mf":"RevK","cu":"http://pongroomac.local/"},"icon":"mdi:thermostat","~":"command/pongroomac","min_temp":16,"max_temp":32,"temp_unit":"C","temp_step":1,"temp_cmd_t":"~/temp","temp_stat_t":"C4D8D5020A31","temp_stat_tpl":"{{value_json.target}}","mode_cmd_t":"~/mode","mode_stat_t":"C4D8D5020A31","mode_stat_tpl":"{{value_json.mode}}","fan_mode_cmd_t":"~/fan","fan_mode_stat_t":"C4D8D5020A31","fan_mode_stat_tpl":"{{value_json.fan}}","swing_mode_cmd_t":"~/swing","swing_mode_stat_t":"C4D8D5020A31","swing_mode_stat_tpl":"{{value_json.swing}}","swing_modes":["off","V"],"pr_mode_cmd_t":"~/preset","pr_mode_stat_t":"C4D8D5020A31","pr_mode_val_tpl":"{{value_json.preset}}","pr_modes":["eco","boost","home"]}
2024-06-29 07:24:54 state/pongroomac {"id":"C4D8D5020A31","up":4,"mqtt-up":1,"app":"Faikin","version":"31e0ca7","build-suffix":"-8266","build":"2024-06-28T23:23:04","flash":64,"rst":1,"mem":37412,"ssid":"Huawei tenet","bssid":"6C5AB05132E6","rssi":-71,"chan":10,"ipv4":"192.168.2.168"}
2024-06-29 07:24:54 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":false,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-29 07:24:54 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:54 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:54 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:55 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:56 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:57 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:58 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:24:59 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:25:00 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:25:00 Faikin/pongroomac {"protocol":"CN_WIRED","ts":"2024-06-28T23:25:00Z","online":true,"heat":false,"power":false,"mode":"C","temp":25.00,"fan":"5","swingv":false,"econo":false,"powerful":false}
2024-06-29 07:25:01 command/pongroomac/mode cool
2024-06-29 07:25:01 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:25:01 info/pongroomac/tx {"protocol":"CN_WIRED","data":"2504500202001050"}
2024-06-29 07:25:02 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-29 07:25:02 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:25:02 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"heat":false,"power":false,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-29 07:25:03 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:25:04 command/pongroomac/mode cool
2024-06-29 07:25:04 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:04 info/pongroomac/tx {"protocol":"CN_WIRED","data":"2504500202001050"}
2024-06-29 07:25:05 state/pongroomac/status {"protocol":"CN_WIRED","online":true,"home":29.0,"heat":false,"power":true,"mode":"C","temp":25.0,"fan":"5","swingv":false,"econo":false,"powerful":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
2024-06-29 07:25:05 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:06 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:07 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:08 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:09 info/pongroomac/rx {"protocol":"CN_WIRED","data":"29C40000000000B0"}
2024-06-29 07:25:10 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
2024-06-29 07:25:11 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
2024-06-29 07:25:12 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
2024-06-29 07:25:13 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
2024-06-29 07:25:14 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
2024-06-29 07:25:15 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000002020000C1"}
[... cut ...]
2024-06-29 07:31:40 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:31:41 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:31:42 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:31:43 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:31:44 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
2024-06-29 07:31:45 info/pongroomac/rx {"protocol":"CN_WIRED","data":"25000012020000D1"}
```

Sonic-Amiga, 2024-06-29:
Oookay... Looks like more complex logic is needed. Sorry, need to think a bit; and have no time this weekend. I'll try to do something about it next week,

ferbulous, 2024-06-29:
Take your time, looking forward to next week

Sonic-Amiga, 2024-07-02:
Hi! Before implementing a straight workaround i decided to try replicating one more aspect of the original Daichi firmware (and we know that original Daikin wired wall control unit behaves similarly). In order not to pollute the repository with perhaps defunct betas, i am attaching the binary here. Please see what happens and if it changes anything.

The idea is that Faikin also sends modes every second, no matter if they chanded or not.

[Faikin-8266.zip](https://github.com/user-attachments/files/16074095/Faikin-8266.zip)

ferbulous, 2024-07-03:
Thanks, will unplug my wemos d1 and flash it again
Can't seem to get beta ota working yet, I even tried used cloudflare tunnel to host the files but I'm still getting error upgrade check

I've unchecked ota beta in the setting
Place the file in ota folder and run the http server from that folder location, then have cloudflare tunnel host the file with this link example. I can view the files just fine in the browser though

https//faikin.mydomain/ota

ferbulous, 2024-07-03:
Great job @Sonic-Amiga 
I think it’s fixed now, no more repeating second command.
Do you still need the logs for it?

Sonic-Amiga, 2024-07-03:
Hi! Happy that the issue is solved. I'll commit it and i think i'll make a release. Thank you for testing.
Please make sure that using remote control also works correctly and Faikin sees changes.

closed 2024-07-03 by Sonic-Amiga

Sonic-Amiga, 2024-07-03:
> then have cloudflare tunnel host the file

cloudflare could do something fancy with redirects etc. Perhaps debug logs could shed some light for you. For me updating works perfectly; i am using it constantly; would be very tedious to re-plug my devboard between CN_WIRED bridge and USB-serial adapter every time.

ferbulous, 2024-07-08:
> Please make sure that using remote control also works correctly and Faikin sees changes.

Hi, that works fine as well, i’ve installed it to all my daikin ac at home 

model -
FTV50PV1L9
FTV28PV1E
FTK50BV1MF

All with CN_WIR ports, I don’t have any model with S21 port. For the wifi built ones, I just disconnect the cables connected to wifi module inside. 

and this also works with Acson AC as reported by other users

ferbulous, 2024-07-08:
Btw, how do I change the LED behavior on the esp8266? For example if I don’t want it to blink

Sonic-Amiga, 2024-07-10:
@ferbulous You can program an unconnected GPIO in advanced settings. Or you can probably just remove all 3 numbers (set to empty); never tried that but logically should work.

Another question - why would you want that ? The LED sits on the board, which sits inside AC case, so at night you don't see it. But blinking LED is a heartbeat, which may offer you some quick diagnostics in case if something goes wrong.

commit 12443a42 (2024-06-28): ESP/main: Repeat CN_WIRED packet 3 times

It turns out some ACs don't hear the command at first time. Original Daichi
controller sends 3 times, so do we now.

Attempts to fix #15
commit 57fee202 (2024-07-03): main/Faikin: Send modes every time as a "response" to CN_WIRED packet.

This better simulates what the original hardware does. Some ACs don't accept
commands in 1st try if we don't do this; perhaps they think we are offline.

Fixes #15 for real, verified by the user.