Sonic-Amiga/ESP8266-Faikin #16: ESP8266 with 1MB of flash booting up only once after flashing and never again

petrica, 2024-07-02:
Hi! I was able to flash the bin files from the ESP/release folder, however the program only boots up once and then never again. Any specific changes needed in order to make it work with 1MB of flash? Thank you!

Sonic-Amiga, 2024-07-02:
Hello! I compile for 8MB; frankly speaking not sure if 2MB is feasible. I know 4MB should be supported because original project supports 4MB and provides partition tables. But never tested.

Sonic-Amiga, 2024-07-02:
Sorry, typo. 1 MB is definitely not feasible; application .bin size is 928 KB; it alone is almost 1 MB. Plus you need size for nvs, bootloader; and a copy of everything (except the bootloader) for OTA.

Perhaps you could make a custom config with a single partition and no OTA. But you're completely on your own here, sorry, major changes would be needed for librevk, which is designed to work with OTA-enabled system.

petrica, 2024-07-03:
Thank you! Since the application boots up successfully I believe the issue is with the nvs partition which is out of bounds for 1MB flash. So, yes, a custom config it is for the partition.

closed 2024-07-03 by petrica

petrica, 2024-07-03:
Manage to make it work with the following partition:

```csv
# Espressif ESP32 Partition Table
# Name,           Type, SubType, Offset,  Size, Flags
RevK,             data, nvs,     0x009000, 0x004000,
otadata,          data, ota,     0x00d000, 0x002000,
phy_init,         data, phy,     0x00f000, 0x001000,
ota_0,            0,    ota_0,   0x010000, 0xec000,
esp_secure_cert,  0x3F, ,        0x0fc000, 0x002000, encrypted
fctry,            data, nvs,     0x0fe000, 0x006000,
nvs,              data, nvs,     0x104000, 0x00f000,
nvs_key,          data, nvs_keys,0x113000, 0x001000,
```

The `ota_0` partition has around 864KB.

To generate the binary for writing the partition with:

```sh
gen_esp32part.py partition-1m.csv partition-1m.bin
```
[partition-1m.csv](https://github.com/user-attachments/files/16084871/partition-1m.csv)

Sonic-Amiga, 2024-07-03:
Thank you very much for your contribution! I will pick up your file; perhaps set up a "size-reduced" build without OTA capability.