Sonic-Amiga/ESP8266-Faikin #17: CN_WIRED status out of sync if Faikin is reset while AC is on

petrica, 2024-07-03:
If I switch on the AC and I reset Faikin, the default settings won't update until I either press any key on the AC remote, or I just start using the Faikin web interface and send commands from there. The issue is that my AC could be on and Faikin would report that it is off.

Is there any way to request a status update from the AC through the CN_WIRED?

Sonic-Amiga, 2024-07-03:
Hello! No, there's no way, that's why Faikin tries to fall back to some defaults.

CN_WIRED conditioners send something every second; but that something can be either sensors or current mode. Your AC chooses sensors; we know that FTN15PV1L sends current modes, and only sometimes sends sensors. That's all we know about the protocol.

ferbulous, 2024-07-10:
Hi, so the faikin unexpectedly restarts/reset when the ac was still on. If this does happen again, is there any indication in the logs why it reboots?

[image]

Sonic-Amiga, 2024-07-10:
How exactly do you see it reboots and not something else ? Uptime ?

This is going to be very tricky unfortunately. Moreover, i am using Faikin myself; and it's rock stable.

What's your exact HW setup ? What kind of 8266 board are you using ?

ferbulous, 2024-07-11:
I assumed the faikin status turning off with the aircon still on was indication that it rebooted and coincidently matches the uptime. Is there another indication that it resets? Checking the uptime just now shows that faikin has different runtime (4 hours+) compared to the 20ac switch turned on (12 hours)

I’m out of wemos d1 so I used an esp8266 dev board that have. I will try it again with a d1 mini and different cables.

Sonic-Amiga, 2024-07-11:
Aha, i see. Indeed that's a restart.
First of all. I checked Wemos specs; and web site says this board is not 5V-tolerant. So you could be frying it on a slow fire.

A crash like this is difficult to catch. A crashing board may only say something on serial port, which is busy. I could produce a special version which dumps logs to serial1, but you'll need yet another device to capture it. Is it okay of a task for you ?

In the meanwhile i'll try picking up fresh changes from ESP32. I know he updated some internal libraries, something could've been fixed.

Sorry, cannot react immediately. I only have 2-3 hours per evening to manage all this. This is complex one.

Sonic-Amiga, 2024-07-11:
After some googling i found out i was wrong regarding 5V tolerance. ESP8266 IS 5V tolerant.

Well; will work through the rest.

Sonic-Amiga, 2024-07-11:
I've just uploaded a new beta, built from very up-to-date code. Picked goodies from ESP32, improved settings and other fixes.
As usual, no time to flash it to my own A/C. Will do on weekend hopefully.

Beta because not tested, sorry. ESP32 on my bench ATM.

LowKey88, 2024-07-12:
@Sonic-Amiga @ferbulous I've uploaded latest beta here : http://faikin-ota.home-assistant.my/beta/

LowKey88, 2024-07-12:
Installed latest beta over OTA Beta: Faikin-8266: aa12406 2024-07-12T00:19:02

Sonic-Amiga, 2024-07-15:
Finally defeated my laziness and installed it on my own AC :)
It required efforts because that one was running ancient version; where OTA didn't work.

Sonic-Amiga, 2024-07-16:
Good news: i think i caught it; uptime on my AC is ridiculously small.
Bad news: have very little time to work on it:(

Sonic-Amiga, 2024-07-16:
@ferbulous I found it. Auto-upgrade is the culprit. Disable it for the time being as a quick remedy.

Version check in librevk doesn't work for 8266, so every time upon bootup it thinks there's an update and reflashes itself. And so on.

This also plays a "nice" joke of downgrading beta, which you might have installed manually, to the latest version, published on revk site (or whatever your update URL is pointing at).

Disabling auto-update should intermittently fix the problem, while i am working on a real fix.

ferbulous, 2024-07-16:
Thanks @Sonic-Amiga 
I was on vacation so I haven't really done any cable/esp replacement
I've just turned off the auto upgrade and that seems to have fixed it (there was also this odd thing happening with the flap just going down and up again every few minutes, i guess this was happening when it tried to reflash itself)

Sonic-Amiga, 2024-07-16:
This bug is quite nasty; so i've just published a release (non-beta), which will disable auto-update for itself. Working on the real fix now.

Sonic-Amiga, 2024-07-18:
I have published a new release; update has been fixed.
In order for the update to work, it's now necessary to store a .desc file alongside the .bin. The .desc file contains versioning information.

New release will also be published on RevK server: https://github.com/revk/ESP32-Faikin/pull/396 . It will be available after the pull request is merged. You'll need to update your Faikin-8266 manually once; then it should be safe to re-enable automatic upgrades.

I am closing this issue as solved.

closed 2024-07-18 by Sonic-Amiga

Sonic-Amiga, 2024-07-21:
Just a note for all readers: Auto-update has been properly fixed; you may now re-enable it.