Sonic-Amiga/ESP8266-Faikin #18: Current Temperature Reading Becomes Null

LowKey88, 2024-07-10:
[image]

**Description:**
There is a bug in the current temperature reading functionality where the temperature reading occasionally becomes null. This issue is particularly noticeable in the Home Assistant interface. The bug disrupts the usability of the temperature reading feature, especially when monitoring multiple units. This issue can be temporarily mitigated by accessing the web settings and making a change, such as updating the OTA URL and clicking save.

Home Assistant Development Tools Output:
```
hvac_modes:
  - auto
  - "off"
  - cool
  - heat
  - dry
  - fan_only
min_temp: 16
max_temp: 32
target_temp_step: 1
fan_modes:
  - auto
  - low
  - medium
  - high
preset_modes:
  - none
  - eco
  - boost
  - home
swing_modes:
  - "off"
  - V
current_temperature: null
temperature: 26
fan_mode: high
preset_mode: home
swing_mode: V
icon: mdi:air-conditioner
friendly_name: Kitchen Aircond
supported_features: 441
```

**Temporary Solution:**
A temporary mitigation for this issue involves accessing the web settings of the ESP8266 device and making a change, such as updating the OTA update URL and clicking save. This action restores the current temperature reading.

Firmware Version: Faikin-8266: 3b55c6b 2024-07-04T23:21:45

Sonic-Amiga, 2024-07-10:
What's protocol of your AC ?

LowKey88, 2024-07-11:
cn_wir/cn_wired both same issue

Sonic-Amiga, 2024-07-11:
Hm, very interesting. Can you send me a debug log ?
By the way, when temperature is reported as null to HA, what does Faikin's own UI show ?

Sonic-Amiga, 2024-07-16:
@LowKey88  Are you still here ?

Disable auto upgrade. It updates in a loop. Version check does not work correctly for 8266.
I am already working on the fix, but it will take time.

Sonic-Amiga, 2024-07-16:
Unless proven otherwise, closing this ticket as being a duplicate of https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/17

LowKey88, 2024-07-21:
Hi @Sonic-Amiga 

I'm still getting this issue, but hard to replicated. Currently I'm using your latest version.

[image]

Faikin-8266: 219028b 2024-07-18T23:59:09 CN_WIRED

Sonic-Amiga, 2024-07-21:
Hmmm...
Well, i need logs. Perhaps some HA interface quirk.

And you didn't answer the question

> when temperature is reported as null to HA, what does Faikin's own UI show ?

LowKey88, 2024-07-23:
[image]
[image]

Yes, seem like the Faikin's UI show the temp : Temps	Home 33℃

LowKey88, 2024-07-23:
I checked on developer tools > States 
[image]

Current Temperature shows null

LowKey88, 2024-07-23:
Here another mqtt log i captured as below, i can see it's can get the temperature home 33 celcius

```

Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T14:16:00Z","online":true,"home":33.00,"heat":false,"power":false,"mode":"C","temp":26.00,"fan":"1","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T14:17:00Z","online":true,"home":33.00,"heat":false,"power":false,"mode":"C","temp":26.00,"fan":"1","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T14:17:22Z","online":0.83,"home":33.00,"heat":false,"power":false,"mode":"A","temp":20.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"1970-01-01T00:01:00Z","online":0.98,"home":33.00,"heat":false,"power":false,"mode":"A","temp":[20.00,23.05,26.00],"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T14:19:16Z","online":true,"home":33.00,"heat":false,"power":false,"mode":"A","temp":26.00,"fan":"A","swingv":false,"econo":false,"powerful":false}

```

Sonic-Amiga, 2024-07-23:
Do i understand it correctly that the log shows 33; web UI also shows 33, but HA shows null ?

Sonic-Amiga, 2024-07-23:
Can you also monitor 'state/<faikin_name>' topic ? Messages look like this:
```
state/faikin32-test {"ts":"2024-07-23T18:35:29Z","id":"08D1F9EE7E8C","up":true,"uptime":4,"mqtt-up":2,"app":"Faikin","version":"7bd5fe4+","build-suffix":"-S1-SOLO","build":"2024-07-23T00:48:54","flash":4194304,"rst":3,"mem":123580,"ssid":"Sonic","bssid":"C4E984287170","rssi":-67,"chan":10,"ipv4":"192.168.0.104","online":true,"power":false,"target":20.00,"temp":23.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
```

Sonic-Amiga, 2024-07-23:
I expect them to be issued every minute.

Sonic-Amiga, 2024-07-23:
This is actually defined by "reporting" setting (specified in seconds). Make sure it's not 0, just in case. For me it's 60 (should be default value).

> such as updating the OTA update URL and clicking save. This action restores the current temperature reading.

Unfortunately that doesn't really tell anything because this particular action reboots Faikin.

Sonic-Amiga, 2024-07-23:
Just in case, re-tested on my 8266 devboard. Yes, messages are sent every minute; i do not change anything on simulated A/C.
```
state/faikin_test {"ts":"2024-07-23T21:31:00Z","id":"58BF25C14773","up":true,"uptime":319,"mqtt-up":316,"app":"Faikin","version":"d344568+","build-suffix":"-8266","build":"2024-07-21T16:39:52","flash":64,"rst":2,"mem":37156,"ssid":"Sonic","bssid":"C4E984287170","rssi":-68,"chan":10,"ipv4":"192.168.0.107","online":true,"power":false,"target":20.00,"temp":22.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
Faikin/faikin_test {"protocol":"CN_WIRED","ts":"2024-07-23T21:32:00Z","online":true,"home":22.00,"heat":false,"power":false,"mode":"A","temp":20.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
state/faikin_test {"ts":"2024-07-23T21:32:00Z","id":"58BF25C14773","up":true,"uptime":379,"mqtt-up":376,"app":"Faikin","version":"d344568+","build-suffix":"-8266","build":"2024-07-21T16:39:52","flash":64,"rst":2,"mem":37156,"ssid":"Sonic","bssid":"C4E984287170","rssi":-68,"chan":10,"ipv4":"192.168.0.107","online":true,"power":false,"target":20.00,"temp":22.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
Faikin/faikin_test {"protocol":"CN_WIRED","ts":"2024-07-23T21:33:00Z","online":true,"home":22.00,"heat":false,"power":false,"mode":"A","temp":20.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
state/faikin_test {"ts":"2024-07-23T21:33:00Z","id":"58BF25C14773","up":true,"uptime":439,"mqtt-up":436,"app":"Faikin","version":"d344568+","build-suffix":"-8266","build":"2024-07-21T16:39:52","flash":64,"rst":2,"mem":35548,"ssid":"Sonic","bssid":"C4E984287170","rssi":-65,"chan":10,"ipv4":"192.168.0.107","online":true,"power":false,"target":20.00,"temp":22.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
```
Could it be HA bug ? Nobody else has reported that.
Would be interesting to see if anything changes in these messages when temperature becomes null.

LowKey88, 2024-07-24:
> Can you also monitor 'state/<faikin_name>' topic ? Messages look like this:
>
> ```
> state/faikin32-test [...]

Yes, it's every minutes

Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T17:09:00Z","online":true,"home":32.00,"heat":false,"power":false,"mode":"A","temp":28.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T17:10:00Z","online":true,"home":32.00,"heat":false,"power":false,"mode":"A","temp":28.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T17:11:00Z","online":true,"home":32.00,"heat":false,"power":false,"mode":"A","temp":28.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T17:12:00Z","online":true,"home":32.00,"heat":false,"power":false,"mode":"A","temp":28.00,"fan":"A","swingv":false,"econo":false,"powerful":false}
Faikin/Living_RoomII_Hall_II {"protocol":"CN_WIRED","ts":"2024-07-23T17:13:00Z","online":true,"home":32.00,"heat":false,"power":false,"mode":"A","temp":28.00,"fan":"A","swingv":false,"econo":false,"powerful":false}

Sonic-Amiga, 2024-07-24:
Stop...

I said the topic should be "state/<faikin-name>", and you show me "Faikin/<faikin-name>"; these are different. What you show me is Faikin's own reporting ("livestatus"), HA doesn't interpred those.

HA message contains "id" (see above).

Sonic-Amiga, 2024-07-26:
@LowKey88 ???

LowKey88, 2024-07-27:
Sorry for the delay in getting back to you; I've been busy with other things. Here are the state logs:

```
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:41:00Z","id":"EC64C9C3D14D","up":true,"uptime":205979,"mqtt-up":205976,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-50,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":30.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:42:01Z","id":"EC64C9C3D14D","up":true,"uptime":206040,"mqtt-up":206037,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-48,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":30.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:43:01Z","id":"EC64C9C3D14D","up":true,"uptime":206100,"mqtt-up":206097,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-49,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":30.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:43:51Z","id":"EC64C9C3D14D","up":true,"uptime":206150,"mqtt-up":206147,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-50,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":31.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:43:53Z","id":"EC64C9C3D14D","up":true,"uptime":206152,"mqtt-up":206149,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-49,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":30.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
state/Living_RoomII_Hall_II {"ts":"2024-07-27T01:44:01Z","id":"EC64C9C3D14D","up":true,"uptime":206160,"mqtt-up":206157,"app":"Faikin","version":"0907519","build-suffix":"-8266","build":"2024-07-21T18:06:13","flash":64,"rst":3,"mem":36872,"ssid":"[REDACTED]","bssid":"6222322A429E","rssi":-49,"chan":1,"ipv4":"[REDACTED]","online":true,"power":false,"target":20.00,"temp":30.00,"mode":"off","fan":"auto","swing":"off","preset":"home"}
```

Sonic-Amiga, 2024-07-27:
This confirms that value is correctly sent to HA ("temp" field).
Could you talk to HA support  and do some debugging on that side ? Unfortunately i'm of no help here, i don't run HA.

LowKey88, 2024-07-27:
Sure, I will check it out.

LowKey88, 2024-07-27:
I forgot to mention that I have one air conditioner connected via the S21 port and I’m using the ESP32 Faikin (https://github.com/revk/ESP32-Faikin). I haven’t encountered any issues with it; it’s been running smoothly since I deployed 4weeks ago.

Sonic-Amiga, 2024-07-27:
I see.
Still there's not enough information. I know what some CN_WIRED conditioners only report temperature periodically. But Faikin hides all those differences. And we see that status messages are sent every minute, and temperature is specified.

Perhaps there's some small bug in HA, which gets triggered by value not changing during some time.  There's definitely some specifics there. Probably Faikin doesn't get something exactly right, but we have no way of knowing it.

BTW, RevK also doesn't use HA, he coded HA protocol just by documentation.

Resume: cooperation with HA is required; any useful info is welcome.

Sonic-Amiga, 2024-07-27:
I will actually try installing HA on my PC and simulating this, but please don't expect that to be done quickly; HA is completely alien for me.

Sonic-Amiga, 2024-08-05:
@LowKey88 Hello!

HA learning curve is very steep for me. Can't find any documentation i need.
In the meanwhile we found a different strange behavior https://github.com/revk/ESP32-Faikin/issues/409 , which apparently depends on "fixstatus" switch. Could you try flipping it ? Also disabling "livestatus" could help (this only concerns Faikin's own status messages, not HA ones)

Apparently they somehow interfere.

MuzzaMcD, 2024-08-12:
> I checked on developer tools > States <img alt="Screenshot 2024-07-23 at 9 52 05 PM" width="849" src="https://private-user-images.githubusercontent.com/1294172/351352527-09e7a442-124b-4523-a699-185b3ed74ca6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NDc0NTIsIm5iZiI6MTcyMzQ0NzE1MiwicGF0aCI6Ii8xMjk0MTcyLzM1MTM1MjUyNy0wOWU3YTQ0Mi0xMjRiLTQ1MjMtYTY5OS0xODViM2VkNzRjYTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTJUMDcxOTEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTQ1ZTFjZWQ1N2U2ZDk5M2M2NzBhYmExNGQxYWI4ZGExMDAzZWUwOWU0YWM4NDViOTcwOTQ1NDIwZjM5Yzk2MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yrhZGgtLZtjuxxa5jVr-54UL7qhiYy4tFNGrS_W4Tds">
> 
> Current Temperature shows null

? Could it be that sending the current temperature of 33 is higher than the value of 'max temp' setting of 32 ?

Sonic-Amiga, 2024-08-12:
Unlikely. max_temp and min_temp, logically, specify allowed range for set point (i. e. target temperature), not what the AC reports. At least i'd expect so. But you may try.