Sonic-Amiga/ESP8266-Faikin #19: Unknown message type Errors

LowKey88, 2024-07-21:
When attempting to subscribe to the MQTT topic `error/Living_RoomII_Hall_II` using `mosquitto_sub`, I am encountering a connection error. The following command is used to subscribe:

```
sh
mosquitto_sub -h localhost -p 1883 -t 'error/#'

error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D000000001092"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D000000001092"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D000000001092"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000010A4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D000000001092"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D000000001092"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"1E7D0000000000A2"}

Receiving repeated “Unknown message type” errors with the following dumps:

	•	“dump”:“1E7D0000000000A2”
	•	“dump”:“00E20000000000B4”
	•	“dump”:“1E7D000000001092”
	•	and others.
```

This only happened to specific aircond that with CN_WIR port. No issue with CN_WIRED port.

Aircond Model is : FTV35PBV1MF and FTKG50T (Both use CN_WIR port)

This was mentioned on this https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/14

It's throw an error on latest build  Faikin-8266: 219028b 2024-07-18T23:59:09

Sonic-Amiga, 2024-07-21:
Hello!
This is actually not an error. This indicates your AC uses a part of protocol that we don't know.
Faikin currently ignores this data.

I wonder, does your AC some extended functionality, not covered by Faikin ? 

Message type is the last digit of the packet (2 and 4 in your cases).
When you use conditioner's own remote, can you make this data (actual body of the packet) to change ? Can we establish some correspondency between these codes and AC functions ?

Sonic-Amiga, 2024-07-21:
> specific aircond that with CN_WIR port. No issue with CN_WIRED port.

Ports are the same :) It's model-specific. This particular model uses protocol extension. It is saying something we don't understand.

Sonic-Amiga, 2024-07-22:
These are especially interesting:
```
	“dump”:“1E7D0000000000A2”
	“dump”:“1E7D000000001092”
```
1E stands for 30. Can you tell about something that was 30 ? Could it be outside temperature ?
Also note flipping from 0 to 1. Can you correlate it with some setting ? What changed  at that moment ?

LowKey88, 2024-07-23:
Yes, it's seem the outside temperature

Here latest log with Faikin's UI screenshot.

error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"207D000000000072"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"217D000000000062"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"207D000000000072"}
error/Living_RoomII_Hall_II/rx {"protocol":"CN_WIRED","error":"Unknown message type","dump":"00E20000000000B4"}

[image]

Sonic-Amiga, 2024-07-23:
Under "extended functionality" i meant some functions on its original remote.
BTW, can you share some info about its original dongle ? What is the chip ? Would be fun if that's ESP :)
In the best case, would be cool to get a schematic, at least displaying how CN_WIR connector is wired to the chip on dongle. Don't know your electronics background. Could you do that ? Or at least make high-res photo of the dongle board on both sides, so that parts and traces are readable; i could try reversing from there.

And is your CN_WIR port 5-pin as shown here ? Is unused pin producing +12V power ?

CheLeh90, 2024-07-24:
Hi, firstly thank you for develop faikin. It makes it easier for me. 

i had same issue. Before this latest update, everything seem okay and no error report.

[image]
[image]
Sometimes, the outside temperature stuck while ac is running, and no cooling time status at the trend.

[image]
This one is another unit look okay with no error report.

i hope it gives you a clue. Thanks

Sonic-Amiga, 2024-07-24:
> i had same issue. Before this latest update, everything seem okay and no error report.

It's not an issue. Just ignore it.
I picked this code up from ESP32 upstream recently. Okay, i'll remove it if it's so confusing. In fact, in the long run it looks like we won't get any knowledge about these extended data packets. In order to discover what they mean, we need some original Daikin HW, which understands them. Original wifi dongle, or wall control panel, whatever. So that i could feed them in and see what happens. But, you know, the world is now split in halves; and i'm not sure if anyone could donate me the required HW for reverse engineering.

Everything i could pull out of import-substituted(tm) "Daichi" controller (from which CN_WIRED was reversed) is already supported.

Sonic-Amiga, 2024-07-26:
I have just made another release, waiting to be published on revk.uk: https://github.com/revk/ESP32-Faikin/pull/405 . Update when merged, the scary message will be gone.

Sonic-Amiga, 2024-07-26:
Note: i switched low fan speed from "Eco" switch to actually a fan speed. Some time ago we agreed with RevK that "eco" was wrong.

LowKey88, 2024-07-27:
No more unknown message types dumped after updating to Faikin-8266: c8f49f5 (2024-07-26T21:39:51) CN_WIRED.

Thank you for the update.

Sonic-Amiga, 2024-07-27:
Good, closing the issue.

closed 2024-07-27 by Sonic-Amiga