Sonic-Amiga/ESP8266-Faikin #22: WeMos D1 Mini 4MB

CPtje, 2024-09-09:
Not familiar with the ESP devices, so please bear with me. Trying to program the files from the release-folder in my WeMos D1 4MB Board using ESP Tool online, but not succeeding.

Is the release suitable for a 4MB Board?

CPtje, 2024-09-10:
Some more info.
Now I tried to flash with:
esptool -p COM4 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition-table-8266.bin 0xd000 ota_data_initial-8266.bin 0x0 Faikin-8266-bootloader.bin 0x10000 Faikin-8266.bin

Unfortunately I get following error:
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: partition-table-8266.bin

Sonic-Amiga, 2024-09-21:
Hello!
Depite the CSV is named "8mb"; the actual firmware image only uses 4MB, and therefore should work fine. I asked RevK, the original author, and he confirmed this. Another user successfully runs it on 4MB.

The error message is weird. Address 0x8000 is just 32668, that's 32 kilobytes.

Sonic-Amiga, 2024-09-21:
Try --flash_size detect . That's totally weird.
I've googled a bit; seems the error could be caused by corrupt files. For example https://esp32.com/viewtopic.php?p=89078&sid=7e1631fe75049659c607811bee80cee0#p89078

Make sure you don't get something like that. partition-table-8266.bin size is 3KB.

Sonic-Amiga, 2024-09-25:
> After using "--flash_size 8MB" it programmed and ran.

Hm, weird, but okay...