Sonic-Amiga/ESP8266-Faikin #23: Strange behaviour after a week of operation

yiannosi, 2024-09-15:
Hi all.
 
 I d like to share my latest experience with Faikin in case someone else has anything similar.

I have installed my FAIKIN on an AC and it has been working pefrectly for more thatn a week.
Just to note that WiFi AP is not always switched on but when it is available, Faikin connects to HA with no problems.

So after a week of normal operation, then blue LED went off (no blinking) and no connection was possible.
I noticed however that during several periods during the day, that lasted less than a minute, the lesty would start blinking again and the FAKIN would connect to the network. A minute later, it goes offline again - LED off. Resting the unit made no difference.

So i had to uninstall the unit from the AC and tried to repeat the above on my bench.
Indeed, hard resetting the esp makes the blue LED blink momentarily once after reset signal is received and no other operation seems possible. I then re-installed the firmware from the serial port which completed successfully but again, after installation the AP would not start as expected and the behavior was exactly the same.
I then suspected a faulty ESP chip so I had it replaced, installed firmware and monitored the unit for a few days.
And again (!) a few days later the new ESP ended up just like the last one.
I tried to reinstalled firmware but the problem was not solved.
Then i decided to install some other kind of firmware. I tried to install a basic ESPHome firmware which was successful.
All WiFi operations are as expected.  Which i suppose means that the hardware itself is ok.
As a last step, I then tried to install Faikin firmare again and - guess what- it now works!
It launches the AP portal, accpets configuration and everything is fine.

So I am left really puzzled
Why re-installing the Faikin firmware didn't solve what ever the problem was?
Why I had to first install another Firmware( EspHome) and **then** install Faikin to make it work?

Anybody met anything similar? Any ideas?

Sonic-Amiga, 2024-09-15:
Hello! Sorry for not responding quickly; i've got infected and a bit sick today, my brain almost doesn't work :)
Could be something wrong with auto update. I had corruption after updating, but fixed it long time ago.

Actually i am successfully running Faikin-8266 in my bedroom conditioner; no problems whatsoever

To narrow things down, let's try to do couple actions:
1. Disable auto update.
2. Check how much flash memory your 8266 has, esptool tells this. My firmware is for 8MB flash.

yiannosi, 2024-09-15:
You mean esptool tells the available  flash during flashing?

Sonic-Amiga, 2024-09-15:
Yes. It also has a dedicated command, "ident" or so... I can't recall, sorry, falling asleep 

⁣ King regards 
​

yiannosi, 2024-09-16:
`esptool -p com5 flash_id`

gives:

```
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:e5:7c:0a:e7:05
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
```

So, i suppose the chip has 4MB of flash. 
However, flashing the firmware didnt give any warnings and it appeared to work fine.
Do you really thing the problem is the flash size?

Sonic-Amiga, 2024-09-16:
Hello! For sure. Attempt to save values or update the firmware goes to nonexistent addresses, wraps around, and kills the program image.
I will make a 4mb build once i recover. Sorry for delay, these two days were nasty.

⁣ King regards 
​

yiannosi, 2024-09-16:
> Hello! For sure. Attempt to save values or update the firmware goes to nonexistent addresses, wraps around, and kills the program image. I will make a 4mb build once i recover. Sorry for delay, these two days were nasty. ⁣ King regards​
> […](#)

You can fit the same firmware in 4MB? Is that possible?

Sonic-Amiga, 2024-09-18:
I'm back, healthy and can hack again.

It is possible. There's plenty of space. There's even a custom 1MB build; but there's no alternative app partition, so no auto-updates.
Moreover, actually the partition table IS for 4MB, despite the suffix says -8m. To my surprise. And there should be completely no problems.
Okay, i'll talk to @revk in the meanwhile... Let's do some testing.

First of all, try turning off auto-update and see if the problem persists.

After the problem reproduces, connect your ESP to computer, set up serial port at 115200 8N1, let's see if the application start. A normally starting Faikin should say something like:
```
<some garbage during init, that's okay>
I (135) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (139) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (146) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (155) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (164) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (174) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (183) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (192) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (201) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (211) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (220) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (229) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (238) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (247) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (257) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (266) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (275) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (284) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0
I (294) RevK: nvs_flash_init
I (297) RevK: nvs_flash_init_partition
I (305) RevK: nvs_open_from_partition
I (333) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (338) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1167.0, 14a6402, Feb 17 2022, 11:32:25, RTOS new
I (395) phy_init: phy ver: 1167_0
I (415) wifi:pm stop
I (419) RevK: WiFi STA [Sonic]
I (421) RevK: Dynamic IP start
I (425) RevK: WiFi STA Start
E (428) RevK: Connect Sonic
I (434) RevK: Up 1, Link down 1 MQTT 0/0
I (447) Faikin: UDP discovery responder start
I (456) Faikin: Trying CN_WIRED
```
After "Trying XXX" the serial port is taken over for A/C communication, and there's no log, that's normal.

yiannosi, 2024-09-19:
The problem is I cannot reproduce the problem on demand!
Remember I had it working for a few days before it appeared.
Anyway, I now have two units. The first one I re-flushed  as explained above and it appears OK but i havent installed it on an AC yet.
The second one, I am not even able to flush it because it doesnt respond at all! Or at least most of the time.

Pressing reset doesnt respond either. I addition measuring the votage at the reset pin when the reset button is not pressed, instead of the expected 3.3V (being pulled high by a 10K resistor) it gives around 0.9-1.2V !!
At times it suddenly starts working, the program runs (LED flashing) and the reset voltage appears normal.  It is as if there is some loose connection, but there isnt!

When it does work, I get the following on the serial port:

`[0;32mI (95) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (104) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (113) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (122) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (131) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (141) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (150) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (159) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (168) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (177) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (187) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (196) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (205) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (214) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (224) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 [0m
[0;32mI (233) RevK: nvs_flash_init[0m
[0;32mI (236) RevK: nvs_flash_init_partition[0m
[0;32mI (244) RevK: nvs_open_from_partition[0m
[0;32mI (284) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
[0;32mI (289) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
phy_version: 1167.0, 14a6402, Feb 17 2022, 11:32:25, RTOS new
[0;32mI (351) phy_init: phy ver: 1167_0[0m
[0;32mI (369) wifi:pm stop
[0m[0;32mI (370) RevK: WiFi STA [Skynet][0m
[0;32mI (372) RevK: Fixed IP 192.168.10.91 GW 192.168.10.254[0m
[0;32mI (375) RevK: Set DNS IP 192.168.10.254[0m
[0;32mI (378) RevK: Set DNS IP 8.8.8.8[0m
[0;31mE (380) RevK: Connect MQTT0 192.168.10.127 as Faikin:MAIN_BEDROOM_AC_704ADF[0m
[0;32mI (387) LWMQTT: Connecting 192.168.10.127:1883[0m
[0;32mI (396) RevK: WiFi STA Start[0m
[0;31mE (399) RevK: Connect Skynet[0m
[0;32mI (405) RevK: Up 1, Link down 0 MQTT 0/0[0m
[0;32mI (424) Faikin: Trying S21[0m`

Sonic-Amiga, 2024-09-19:
I soldered an smt board once by hands; and the name for this behavior is "flux". Sometimes they only say it doesn't require washing; it does. And it perfectly sticks around for quite a long time under flat parts like transistors, chips and ESP module :) I recall i had to remove a PIC controller and even quartz and clean the mess thoroughly before things worked.

Your serial log is perfectly normal 

⁣ King regards 
​

Sonic-Amiga, 2024-09-19:
If happens again , you can also dump flash (whole 4mb in one block) according to instructions http://stackoverflow.com/questions/53920782/ddg#53924654 and mail me.

Using this dump i can verify corruption hypothesis.

⁣ King regards 
​

yiannosi, 2024-09-19:
"I soldered an smt board once by hands; and the name for this behavior is "flux"..."

Yes I know . It happened  to me tool But this is not the case here. I made sure that coon3cti9ns are ok.

yiannosi, 2024-09-19:
"Using this dump i can verify corruption hypothesis."

will do..
In the meantime, pls tell me when you have a 4mb version

Sonic-Amiga, 2024-09-20:
The version you're using is 4MB. There's no real 8mb version. Confirmed by RevK.

yiannosi, 2024-09-21:
ok.. Seems that like the ESP module was somehow faulty. Had it replaced and now everything looks ok.
I have installed the Faikin on the AC today and so far works as expected...

closed 2024-09-21 by yiannosi