Sonic-Amiga/ESP8266-Faikin #24: CN-Wired connection

yiannosi, 2024-10-15:
Anyone has any photos of the Daikin PCB where the Faikin is connected for a ducted system?
Where exactly is the S21 port located in this case?

By the way, ducted units also have a wall mounted wired controller  which seems to communicate with the main unit with a two wire - probably RS485 protocol. Has anyone ever tried to connect to this cable?

[image]

Sonic-Amiga, 2024-10-21:
Hi there! Have you received my email?
 Ducted units use different protocol and different connector called x50a. It only provides data, the power is another connector, i cannot recall how it's called. I am on Bali on vacation

yiannosi, 2024-10-21:
Hi yes!
 Ive been trying all weekend to get it to work with no luck.
Used X50A along with X35 for power.  Faikin power up but remains offline with x50 protocol selected. Maybe X50a has different pinout or it's disabled in my case...
Seen another similar post..

Sonic-Amiga, 2024-10-21:
I have an x50 cable which came with BRP069C42 back at home. When i return, i will check the pinout

⁣ King regards 
​

Sonic-Amiga, 2024-11-19:
Just to mark this complete. The pinout of my cable is here: https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/11#issuecomment-2454487473

But, just two wires is neither of our 3 supported protocols. From another thread we see it's P1P2; and sorry, it needs quite different hardware: https://github.com/Arnold-n/P1P2MQTT P1P2 uses differential signalling plus it's capable of delivering power over the same 2 wires.

closed 2024-11-19 by Sonic-Amiga